ten-point-nine/freETarget #1: The display part

modena01, 2020-03-03:
Hi Allan, what calculation output do you intend the Arduino to output? Just thinking about the display code, will the Arduino output X/Y coordinates (in mm?) or will it output raw timers values, or something else?

ten-point-nine, 2020-03-03:
The Arduino will read the counters, perform the calculations, and output the position as a JSON message

The display PC would have to make a pretty picture

Allan

ten-point-nine, 2020-03-03:
I figure an xy coordinate in mm would do nicely

That way pistol or rifle should work.

I’ll put together an ICD once I send the board out for fabrication 

Allan

modena01, 2020-03-03:
Thanks, easy.

modena01, 2020-03-06:
will the Arduino be outputting to an IP network interface, or are you planing on serial or something else?

ten-point-nine, 2020-03-06:
I was thinking serial over usb.  But no reason not to be WiFi

Allan

modena01, 2020-03-06:
WiFi be best, yes.

TigraTT, 2020-08-12:
as far as i know this project should work for private ranges, but i think that some shooters want to have a cheap solution for ther club ranges.
I and some of my teammates for example want to build up 5 or 10 ranges for 50m and 100m .22 rifle shooting.
For this i thought about a 5 port gateway from serial to ethernet like Moxa or Advantech.
I can check such an solution here in my office because i´m working in the industrial IT and have these parts here for testing.
I will check this and give you an feedback.

AzmodanRO, 2020-08-12:
well, Allan would probably know better...
but as far as i know, we are limited on what the arduino board is capable. 
and the current arduino only supports USB for direct connectivity. There might be solutions to extend the USB over longer distances via ethernet or wifi, but that would only be a bridge.

closed 2021-01-31 by AzmodanRO