ten-point-nine/freETarget #2: Missing what the sensors are and so on, how can i contact the author?

ArnoBrinkman, 2020-03-21:


ten-point-nine, 2020-03-22:
Hi

Sorry with all the crisis going around your email fell down my list.

The sensors are four microphones in a circle.   Currently working on the microphones now.

Can you help with this?

ArnoBrinkman, 2020-03-22:
I couldn't find your mail address or other way to contact you then this.
My mail address is arno@abvisie.nl

ten-point-nine, 2020-03-22:
Thank you

I can be reached at allan.brown.atl@gmail.com

How did you find the github page?

Regards

Allan

clivegarnham, 2020-03-23:
Hi Allan, 

Have you thought yet about what type of microphones to use? Do you have any idea what other target systems use? I would be willing to do some experimental and development work in this area if that would help.

I see from your documentation that your interest is making something for 10m air, and that seems a great place to start. I would be interested in extending the system to include smallbore distances, ie 25 yards/metres, 50 metres and even 100 yards. With different target hardware, I think the same basic system could be adapted to work at all these distances. For competition use, I'm not sure that the counter size and clock frequency are high enough, but that is something that can be looked at later.

Regards,

Clive Garnham

ArnoBrinkman, 2020-03-23:
I've send you a private mail

ten-point-nine, 2020-03-23:
I have not done much work in that area, so any help would be great.

The cable harness supplies 3.3 and 5.0 volts.  There are separate conductors for each microphone. 

The circuit goes into a simple competitor that has an adjustable reference voltage.   I anticipate a trip voltage of about a volt would be ok.  Once started the software shuts down the circuit after 500us

I was thinking of an wideband microphone to capture the rising edge of the sound wave

Anything you could do would be great

Thanks

Allan

closed 2021-08-14 by AzmodanRO