ten-point-nine/freETarget #4: How make to start de software on Windows

Rebelmuzick, 2020-06-12:
Hello friends, I am a member of the facebook group of 10.9, I was interested in putting together this electronic target, my question is how can I run the software with windows 7 or 10. What software do I need to run the program in C #
Thank you so much

AzmodanRO, 2020-06-12:
you need Visual Studio 2019 Community Edition  (it's free - download from
microsoft)

Rebelmuzick, 2020-06-12:
> necesitas Visual Studio 2019 Community Edition (es gratis - descarga desde microsoft)
> [...](#)

Thank you very much, surely when you have the microphones of the scheme and build the boards I will consult how they are mounted and then how to program the various sketches of the arduino since the times that I have done it I only programmed one sketch. ino at the same time. total thanks

closed 2020-06-12 by Rebelmuzick