ten-point-nine/freETarget #5: Changing PC Software to work an 25m Pistol and 50m rifle

TigraTT, 2020-08-12:
dear Azmodan,
i did not know if we are in contact before, but i´m in contact with Allann over messanger.
And i´m sorry, but i could not reach you ofer the Meisterschützen Forum or any other way, so i post my question here.

at first a big thank your for giving us this freeware, i think your project could be a very good solution for all shooters to get a cheap electronic target.

I chat with Allan for ordering two sets of electronic boards, but actual they have a longer delivery time. If i understand him right??

You have to know that my skills in programming are not so good, and C# is completely new for me. I am working in industrial automation and there i know several PLC systems and a little bit VB.net.

I have loaded the software and i could change for example the air pistol target to a 25m .22 Pistol target. (in Germany we use the same target for smallbore rifle at 100m)

Also i have changed the zoom so if i change between Pistol and rifle, the targets are both fullscreen.

My Problem is, if i make some shots (with the simulation) i get the right picture but the shot value is wrong. (On the 25m target!)

For example a shot in "Center" is right with 10.9, but a shot with -30 and 0 is a 7,2 instead of a 9,8?

I understand that the value is calculated for the 10m air pistol target, but i do not find this part in your program to change this calculation offset to my new target.

(30mm on the ar pistol arget seems to be a 7,2)

I think if you can add a 25m Pistol target and also a 50m rifle target this would bring a great boost for your free Target and you can sell many of the boards.

as i read in Facebook oder some shooters forums this question is coming up from different users?

I want to build up two completely working targets to show them in my club, to convince other shooters to invest in 8 more targets, because then we have all our 50m and 100m lines running electronic targets. If we want to buy them at Meyton, Sius, Seta or others, this will cost ust about 50.000€ and this is money we do not have.

Did you test for example with USB to Ethernet adapters to get a longer distance than 10m??

Kind regards, and thanks a lot for your free work!

Ralph

AzmodanRO, 2020-08-12:
hi
yeah..i don;t check the meisterschutzen forum very often :) here is better.
yes, you are correct, that the 25m and 50m requests are abundant. we are looking at this for sure, but first we would like to finalize the current setup (10m) since it;s easier to test (at home - for longer distances we need a shooting range). 

please understand that the project is still in beta testing and the time frame for further features is still uncertain. Allan also has to perfect the fabrication process. he is doing it manually as of now.

but, in the interest of the current conversation, i would need exact specification of your 25m target. is it the free pistol target, used for 25m center fire precision stage also? since i only shoot 10 air pistol, i;m not too familiar with small bore targets/distances/disciplines

TigraTT, 2020-08-12:
hi, and thanks for your answer!
i do understand that this is a beta, and i know that time is the most expensive material in such projects.
At the moment me and my teammate (he is Carpenter) are working on a wooden target and the cabeling for about 100m.
This is not easy because 100m is a bit too long for RS-232, so i decide to work with serial to ethernet converter, or maybe to a RS232 to RS485 transmission. I will give you a feedback if it is working.
On my private Meyton frame i have a RS232 connection, but this is only 50m!

i have played around a little bit and i found these settings in your code:
**public const float pistol1X = 5.6f; //oficial scoring dimentions
public const float pistol1Y = 10.3f;
public const float pistol2X = 4f;
public const float pistol2Y = 10.5f;**

if i change the 5.6 for example to 10 the value of the shot is calculated better but not right.
But if i change this, a 10.9 is not longer a 10,9, so i wonder where i have to change more??

here is the code for the 25m target (or in your words the Free Pistol 50m target):
public const decimal outterRingPistol = 500m; //mm
public const decimal ring2Pistol = 450m; //mm
public const decimal ring3Pistol = 400m; //mm
public const decimal ring4Pistol = 350m; //mm
public const decimal ring5Pistol = 300m; //mm
public const decimal ring6Pistol = 250m; //mm
public const decimal ring7Pistol = 200m; //mm
public const decimal ring8Pistol = 150m; //mm
public const decimal ring9Pistol = 100m; //mm
public const decimal ring10Pistol = 50m; //mm
public const decimal innerRingPistol = 25m; //mm

public const decimal innerTenRadiusPistol = 25m;

the air pistol target has ring wide of 8mm and the 25m target has ring wide of 25mm, and the pellet caliber is different from 4.5mm air pistol to 5.6mm Free Pistol.

did you have any informations when the boards will be deliverable or is this Allans part?

Kind regards, Ralph

AzmodanRO, 2020-08-12:
as i wrote in the a code comment, i use linear interpolation using the calculations from here: http://targettalk.org/viewtopic.php?p=100591#p100591
i have no idea how he calculated those values (decimal score versus shot radius ), but they work. if i can find the formula, i will implemented in the code so no to have hardcoded values

TigraTT, 2020-08-12:
i have found it and changed your code to his values and it seems to be working!
 **public const float pistol1X = 16.35f; //5.6f; //oficial scoring dimentions
        public const float pistol1Y = 10.4f;  //10.3f;
        public const float pistol2X = 13.85f;  //4f;
        public const float pistol2Y = 10.5f;**

i will continue with testing and i hope that i can get two sets of boards in the near future.

AzmodanRO, 2020-08-12:
cool
i am also waiting to get a sets of boards :)
after we are happy with the current setup, i will refactor the code to include more targets

TigraTT, 2020-08-12:
these are very good news!
if you need help, for testing on ranges up to 100m, please contact me.
i also will try to improve my skills in C#, maybe i can make some easy changes, such as translation for example.
I have even made the translations for the Trace! shooting software.

ten-point-nine, 2020-08-12:
Just joining at the end

I added a calibration message to my Arduino code.  

How big is a 100 meter target 1 ring?

Allan

ten-point-nine, 2020-08-12:
I modified the Arduino code so that you can calibrate it for a larger target.

The JSON message {"DIA":new size} **to** the Arduino will adjust the algorithm for the new sensor location.

The sensors must be placed at least 1.4x the diameter apart.  

Right now I seem to have a problem with shots that are close to one sensor and far away from the other (ex 3 ring at 12'oclock)  I think if I boost up the microphone sensitivity I should be OK.

As Azmodan  said earlier, right now time is the most expensive commodity, and with the summer and COVID, things are not moving as fast as I would hope.

TigraTT, 2020-08-12:
Hi together,
i have send the code with the target size in my second post, but here is a picture. (only in german, but i think it is the international pistol precision or free pistol 50m target.
[image]

i have tested with the targetsimulator and the shots seems to be right.
So if you trasnmitt the distance in mm from center, the shots are right on target and value.
As far as i could see, the protocol transmittes the X/Y distance, the (square root of X²+Y²) and also the degrees for the arrow, and Azmodan is then drawing the target out of these strings.

a question to the sensitivy of the microphones, did you "hear" the pellet hit the target or did you "hear" the ultrasonic fron flight through? Maybe it easier to hear the hit of a target??

have a nice evening or day,
Ralph

ten-point-nine, 2020-08-12:
For the moment the microphones are picking up the 'pop' as the pellet passes the microphone

I want to avoid the sound of hitting the target since that would need to add in a motor and witness paper.

TigraTT, 2020-08-13:
OK, i understand this issue.
but did the microphones are not disturbed by other sounds or noises around?
For example if you have two or more shooting ranges?
I have made a test with the emulator and "shot" every 25mm to see if the calculation and also the drawing of the shot was OK.
It seems that everything is OK so far.
[image]
My teammate is going on with carpenting a target and if we get the electronic boards we will continue with testing.
I hope we can manage it all together to get a proper shooting range.

If you ever come to germany, we would be glad that you visit us at our range to have some steaks and beer!
"or Tofu and raspberry lemonade ;-)"

TigraTT, 2020-08-13:
This is the code for 50m ISSF Rifle Target:

        // changed to 50m rifle .22 5,6mm
        public const decimal outterRingRifle = 154.4m; //mm
        public const decimal ring2Rifle = 138.4m; //mm
        public const decimal ring3Rifle = 122.4m; //mm
        public const decimal ring4Rifle = 106.4m; //mm
        public const decimal ring5Rifle = 90.4m; //mm
        public const decimal ring6Rifle = 74.4m; //mm
        public const decimal ring7Rifle = 58.4m; //mm
        public const decimal ring8Rifle = 42.4m; //mm
        public const decimal ring9Rifle = 26.4m; //mm
        public const decimal ring10Rifle = 10.4m; //mm

        public const decimal innerTenRadiusRifle = 5.0m;

        public const float rifle1X = 5.60f; //oficial scoring dimensions
        public const float rifle1Y = 10.3f;
        public const float rifle2X = 4.8f;
        public const float rifle2Y = 10.4f;

there is only the problem that the 10 is completely white, but is has to black with an inner circle like pistol target.
i will try to find it and change it.

ten-point-nine, 2020-08-13:
TigraTT

I do not have an answer about the cross talk issue on the target.  I was hoping to get some data from the field.  I am planning Version 3 of the board (Currently at 2.2) and might put a fifth microphone in to listen for loud noises and squelch the input for a few milliseconds.

Prior to the COVID-19 I periodically worked in Wurzburg, Stuttgart and Villingen.  If travel ever opens up again I will take you up on your offer.

TigraTT, 2020-08-13:
OK, Würzburg is about 240km from our Range, but it´s too far for "short visit" i think, Stuttgart is even further away.
In the meantime i will wait patiently for getting some of your boards.

TigraTT, 2020-08-17:
i have made some more changes, and i think you will see then in my picture.
The changes are only for having a better sight for people like me who needs glasses to read the small desktop fonts.
Even when you are on the range, you did not wear your reading glasses so some fonts are very small.
I did only a fontsize change for the last shot and for the shots summary.
But as you see in my screenshots, there is a problem in the table when the total of one target reaches 100,x, then the field is too small. I have made it larger to display the complete string, and for that i have decreased the first column. But now i could not get the complete text for T1 in the firrst column. 
Even the position arrow is not going to the next target row, it stays in row 1.
Did you have an idea where to change this?

[image]

AzmodanRO, 2020-08-17:
yeah..that grid control is pretty finicky. i struggled a bit to make it look as it does now...
i'll see what can be done to increase the font size

TigraTT, 2020-08-17:
i think the fontsize at all is not that important. The two changes for Last Shot and Total shots are OK (for me). Windage, Spread and so on are a less important when shooting. And after shooting you can put on your glasses and see everything fine.

For your intereset:
I have testet with a USB to Ethernet Hub where i connected my USB to Serial adapter that works fine directly on my pc, but on the Ethernet adaptor it is only transmitting the first shot OK and at the second shot the freeTarget software is crashing.
In case i have checked with two different USB to Serial adaptors, i think i will work with an active RS232 serial extender for the 50m.

AzmodanRO, 2020-08-17:
please post a stacktrace and/or screenshot of the error

AzmodanRO, 2020-08-17:
posted a change to make the target breakdown total bigger (to fit 109.0)

TigraTT, 2020-08-18:
hi Azmodan, the table now shows only full rings at pistol setting and tenth of a ring at rifle settings.
i also get an error when changing from practice to match when i stay connected to COM port, if i disconnect and change between practice and match, there is no error.
This is the error:
System.DllNotFoundException: "Die DLL "SQLite.Interop.dll": Das angegebene Modul wurde nicht gefunden. (Ausnahme von HRESULT: 0x8007007E) kann nicht geladen werden."

i´m sorry, but my skills are not sufficient to correct the bug.

AzmodanRO, 2020-08-18:
your error is because you run the program from visual studio. for that, you need to install to external packages: SQLite and PDFSharp (with the NuGet tool from inside visual studio). check the readme.txt in the freETarget main directory

TigraTT, 2020-08-18:
OK; Understand. But this error is coming up in the older version.
I had an older SQL version in my studio. after upgrading everything is fine!
Thanks a lot!
i have seen that there is also a column to show the serial ports data exchange. Did you think that in a final version this column can be hidden to get more space for the left side information column, because i think then we can increase the fonts to see them much better than now.
i know that i have a lot ideas, but please understand that i came from industrial visualization and SCADA software, so my first thought is always the information for the user. Because of this, i think the shooter did not need the serial port informations during a practise. I also know that this project is only for hobby enthusiaths and not for bigger envents or championships.

Please don't get me wrong, i'm totally excited about the software and the project as itself, and i will make some advertising so that you can sell as much systems as possible! And i´m totally happy to be a small part of this project, it gives me lot of fun to learn more and more about it.

AzmodanRO, 2020-08-18:
the console can be hidden from the preferences window , uncheck the "display debug console".

about the space... the problem with most UIs, is that they are drawn in raster mode (by pixels), not in vector mode. so to make it look good, you have to work with fixed and precise dimensions. if you go with vector UI, it will be very scalable, but it will look very ugly.

the list on the left of the target and the target breakdown on the right, are raster UI controls - that behave very bad when you try to resize them (scroll bars appear, text disappear, lines overlap or break apart and so on). that is why the left and right side are fixed in dimension (width), while the central target (that is drawn in vector mode) is scalable. 

i understand the need for clarity, but when working with normal PCs and laptops, they have a lot of dimensions ,resolutions and DPIs that make it impossible to get a fixed look - unlike industrial software, where the vendor sells the display unit also, so he knows exactly on what display the software will be shown and it will never change.

TigraTT, 2020-08-18:
i agree, the different resolutions and formats are also a big problem i have to fight for every day with our scada software here in the company when customers change their hardware.
Your solution is working fine, so i do think nobody needs a vector graphic based solution.
at the moment i´m very happy with the results o my test and i´m awaiting the boards to go further with testing a real measuring frame on our range. In case i have a private Meyton frame, i can put them together for testing the recognition of our diy frame.
I hope the boards are available soon............

TigraTT, 2020-08-28:
at least i have changed a little bit for the 25m pistol target, there it was possible to get a "11,0! instead of 10.9.
Now after a small change this ist fixed now.
I have copied the code from the rifle target, because there it was implemented before.

@ten-point-nine:
Are there any news about manufacturing the boards?

TigraTT, 2020-09-09:
i have made some more changes to the user interface.
The 10 ring shots now are shown in red, 9 ring shots in yellow and lower than 9.0 are schown in lightskyblue.
I made this modification because this is better known for the Meyton users and also because of the loosing of eyesight when getting older. Now you can tell from the colors if it is a 10 or 9 or less than 9.
I think its a nice to have.

AzmodanRO, 2020-09-09:
good idea. i;ll integrate it in the main sourcecode

TigraTT, 2020-09-09:
here is my code:
if (l == count - 1) {
                if (shot.decimalScore > 9.9m)
                {
                    c = Color.Red;
                }
                else if(shot.decimalScore > 8.9m && shot.decimalScore < 10.0m)
                {
                    c = Color.Yellow;
                }
                else if (shot.decimalScore < 9.0m)
                {
                    c = Color.LightSkyBlue;
                }

                p = new Pen(Color.Blue);
                bText = new SolidBrush(Color.Blue);
            }

AzmodanRO, 2020-09-09:
so you want the old shots (not the current one colored)? not the current one (last shot)?

TigraTT, 2020-09-09:
OK, i will try to explain with my "broken englisch"
the actual shot after my changes now is red or yellow or skylightblue.
The older shots are now in blue, and these older shots i want to make grey. This because of two reasons, first one is that grey is better visible on black than blue and second because then only the new shot has a "colour" and the old one does not stand out so strong appearing to black. I will make some tests with different (50) shades of grey.
The option with inner tens is an idea of me, and i want to check out if this works fine.
Even i have changed the fontsize of the shots from caliber /3 to caliber /2, because the .22 bullet is bigger than the air rifle diabolo.

ddraig576, 2020-09-09:
I’d suggest checking this for compatibility in case anyone has colour blindness. 

https://usabilla.com/blog/how-to-design-for-color-blindness/

Although I’d hope Meyton had already dealt with it if following their colour scheme

TigraTT, 2020-09-09:
Meyton has a green background of the target and draw the shots in red, yellow and blue.
I don´t know if they have made it regarding some rules for colour blindness, but i think the colours depends on the very old systems of Meyton running on very poor hardware with maybe only 16 clours?? (Thats only my idea, i do not have any informations about this)

TigraTT, 2020-09-09:
here is a sample how i have managed it so far.
I hope you can see what i thought about.
The only thing left to do for me is painting the older shots in grey instead of blue.
[image]

TigraTT, 2020-09-09:
now at least here is my latest code and a screenshot:
//draw shot on target
            int count = getShotList().Count;
            Color c = Color.FromArgb(200, 190, 190, 190); //semitransparent shots
            Pen p = new Pen(Color.LightSkyBlue);
            Brush bText = new SolidBrush(Color.Black);
            if (l == count - 1)
            {
                if (shot.decimalScore > 9.9m)
                {
                    c = Color.Red;
                }
                else if (shot.decimalScore > 8.9m && shot.decimalScore < 10.0m)
                {
                    c = Color.Yellow;
                }
                else if (shot.decimalScore < 9.0m)
                {
                    c = Color.LightSkyBlue;
                }

                p = new Pen(Color.Blue);

                if (shot.decimalScore > 10.2m)
                {
                    bText = new SolidBrush(Color.Yellow);
                }
                else
                    bText = new SolidBrush(Color.Black);
            }

[image]

ten-point-nine, 2020-09-09:
I would like to add a feature to the UI

Download a firmware image from the PC to the Arduino.

Needs

A dialog box to locate the Arduino file
Open a serial port at 38,800
Send some control bytes to the Arduino 
Send the contents if the file to the Arduino 
Close the connection
Close the file

If I get the details of the flow can one of you code it?

Allan

TigraTT, 2020-09-09:
Allan, sorry but thats too much for my poor skills in C#
I´m very happy that i could change the things i have made so far, and mostly happy that they are working!

AzmodanRO, 2020-09-09:
of course i can code it

AzmodanRO, 2020-09-09:
pushed the shot color customization to git

TigraTT, 2020-09-09:
i found that you have changed the fontcolour for all 10ring shots, i only have made Yellow for "Inner Tens", means over 10.2

AzmodanRO, 2020-09-09:
inner ten means different decimal score depending on the target. for pistol it;s above 10.4

TigraTT, 2020-09-09:
OK, thas right.

TigraTT, 2020-09-09:
with the latest download i get an SQL Lite error when debugging??
Sorry, but i did not know how to continue

AzmodanRO, 2020-09-09:
what error?

TigraTT, 2020-09-09:
this is the error:
[image]

AzmodanRO, 2020-09-09:
yes. i added that check to make sure the journal database is accessible.
if you are running the program from visual studio, u need to install the "System.Data.SQLite" package using NuGet

TigraTT, 2020-09-09:
OK i found it!
Nice work with the settings page for custom colours!
Now there should be only one more color for "Inner Tens" text colour and maybe for the fontsize?

AzmodanRO, 2020-09-09:
i intend to refactor a lot of the "target" setup in order to support all target sizes. in there will be the place to setup more stuff like font size and the like

TigraTT, 2020-09-09:
you are my hero!
i think this project will grow and make many shooters happy with a cheap and simple electronic target for practice and training.

I have so many ideas to implement in such a training system, for example to work online with a Scatt or Trace! sensor, building time axles for analysing the shot rythm and so on.

A next step for evolution could be a LED lightning frame that could be controlled in brightnes and light colour by the user interface?
I have build a light frame for my training at home with a Trace! sensor, but the brightness is only adjustable by a poti.
[image]

ten-point-nine, 2020-09-09:
There was a delay in getting production going so I took the opportunity to add an accessory connector to the board.

It provides power and ground and 4 logic/communication signals.  The intention is to drive a paper roll or change from USB to something like RS-488

It would be easy to have a light driver to illuminate the target.

Allan

mjbedford, 2020-09-09:
On the topic of external connections, will the Arduino shield allow for stacking shields?

I’d like to explore using a messaging protocol over ethernet or wifi, and would like to into face with a DIP switch or similar to set an identity for each target so each lane can be identified individually output captured mentally for competition scoring etc…

Martyn

ten-point-nine, 2020-09-09:
At the present there is no plan to stack shields. 

There is a jumper heading on the board that could be used as a lane selector.  There is also provision to download an electronic DIP switch setting via JSON but currently there is no software that uses it

The accessory connector can be used to drive an external Ethernet or WiFi link if the software was added

Allan

AzmodanRO, 2020-09-09:
multiple targets should be linked at the PC software level...

mjbedford, 2020-09-09:
Doing it at the PC software level would be a real bottleneck and make it very dependent upon a complex solution, using a pub sub message protocol you can decouple the target, the display and the aggregator. This means that you can connect displays to a message bus, dial in the lane number and all the connectivity is solved. It also means that you are not dependent on one PC solution for displaying shots, you could use phone, tablets etc...

mjbedford, 2020-09-09:
Sorry I haven't looked recently at the Arduino code, what protocol are you using between the shield and the mega, I2C or SPI? an ethernet shield could sit between the shield and the mega, using I2C or SPI should mean they don't interfere with each other.

mjbedford, 2020-09-09:
I like the idea of integration with SCATT etc... have you thought about how that would work, SCATT is a very closed system with no interface I can find to access data or input data, we have only a SCATT basic which is great at recording traces but really struggles to place the shot accurately. It would be great to get it to listen to the target to adjust the fall of shot to what the freETarget reports.

> you are my hero!
> i think this project will grow and make many shooters happy with a cheap and simple electronic target for practice and training.
>
> I have so many ideas to implement in such a [...]

ten-point-nine, 2020-09-09:
My board uses about 40 GPIOs to access the data acquisition hardware. 

Allan

mjbedford, 2020-09-09:
SPI on the mega uses pins: 50 (MISO), 51 (MOSI), 52 (SCK), 53 (SS) I seem to remember that they may be re addressable if in use, are you using 50 through 51?

ten-point-nine, 2020-09-09:
The accessory connector will have two serial ports plus power and ground. The serial ports can be configured as discretes if needed

The SPI port is used to access the temperature probe

Allan

TigraTT, 2020-09-09:
i do not have any idea how to combine Scatt or Trace! to freeTarget, but i know that there is a former german 1. Leage shooter who has developed it with a Scatt (Pro or Expert) and Meyton frames.
He also developed some other shooting software, his website is do4you.ch
But as far as i know this is not freeware, and i also did not know how far he has developed it.
A few month ago i have contact to him because the Trace! developers are interested in a collaboration and i know one the Trace! developers from a earlier work were i have translated the Trace! software to german.
The Trace! software is downloadable for free, but it only runs with a sensor plugged into USB port.
If anyone is interested in a stored traning session i can provide a Trace! session file.

AzmodanRO, 2020-09-09:
i also have a trace sensor. it would be nice to link them.
but first things first. let;s make sure basic functionality is working. (10m shooting). when that is working smoothly, Allan has the production pipeline running and more people are using the system, we;ll talk about new features.

[2 more replies]

closed 2021-01-31 by AzmodanRO