ten-point-nine/freETarget #11: Sensor board dimensions and placing

mjbedford, 2021-01-31:
Continuing the thread on FB.

I am modelling a frame for the sensors to be attached to the back of a faceplate, and a gig to ensure it is exactly centred on the target holder, to get the distance exactly correct I would like to know if the 230mm distance is from the face of the sensor mount or the front of the microphone. If it is from the microphone could measure the height of the microphone and the PCB thickness. Thanks Martyn

AzmodanRO, 2021-01-31:
the distance from the microphones from face should be 230mm. the PCB plus microphone has 4mm in thickness, but i put some distancing washers (so that the soldered pins on the back of the PCB are not crushed). so the final "thickness" of the sensor is 6mm. that means that the sensor block should be 6mm away from the point where the microphone face will be.

i took that into consideration when i built my 3D model

[image]

[image]

AzmodanRO, 2021-01-31:
[image]
[image]
[image]
[image]

ten-point-nine, 2021-01-31:
Hello Martyn

The 230 mm is measured from the face of the microphone to the opposite face.

The 230 is the 160mm target dia x sqrt(2)

Allan

ten-point-nine, 2021-01-31:
Let me see if I can get the DXF of the board.

Version 3 sensor boards will be the same size.

Allan

ddraig576, 2021-01-31:
I was thinking a 3D printed mount for the modules to go in the corners of something similar to a SCATT LED light box (I'm looking at possibility of getting a cheaper model manufactured locally) would work well for 10m. Obviously for .22LR or high power air, the target face would need to be thicker steel to protect the components. 

Doug

mjbedford, 2021-01-31:
Thats great Thanks, I have a parametric model so I can tweak the dimensions based on experiment easily enough without having to rebuild
I'll post a render when I have made a bit more progress

mjbedford, 2021-01-31:
[image]
A rough render, the top layer is a single print with the sensor mounts as one piece, I have also modelled a jig which fits in a 155mm hole in the front plate of your choice and aligns the sensor mounting frame to the back a bit of glue, screws or your preferred brand of duck tape and you should have the sensor in the right place and you can remove the jig.

I'll run a print off during the week to test for alignment and tolerances etc..

mjbedford, 2021-01-31:
P.S. its small enough to print on a domestic 3d printer and should cost no more than $2 to print the frame and the same for the jig

ten-point-nine, 2021-01-31:
Martyn

The hole in the middle needs to be square and the corners as close to the sensors as possible.

During test I found occasions where the circle could occlude the sound from reaching the sensor.

Allan

ten-point-nine, 2021-01-31:
How are you attaching the sensor to the printed plate?

Allan

ten-point-nine, 2021-01-31:
PS

Looks like you’re on the development team. I can bump you up to the top of the list.

I can send you a PayPal request if you like.

Allan

AzmodanRO, 2021-01-31:
the target face needs to look something like this:
-square hole (16x16 cm for air pistol)
-mounting blocks for the sensors at the corners of the hole (230 mm for air pistol)
-some way to put the paper target on the other side of the plate, with the position on the paper aligned the hole and the sensors (the center of the paper target needs to be at the center of the hole and at the intersection of the sensor diagonals)

[image]
[image]

ten-point-nine, 2021-01-31:
As to tolerances.

Moving one of the sensors by 1mm radially will move the centre pint by 1/2 mm.  It would have a gain effect over the span of the target.

Centered on the paper target is nice, but a bit like SCATT the gun is sighted in to what the target thinks is zero.  This gives a bit of tolerance to the overall construction.

I use regular paper targets because I am too cheap to buy the good ones

Alla

niggli, 2021-01-31:
I am also going to construct parts for 3D printing as soon as I have the hardware, but I would suggest not to integrate the front plate: 1) ist going to make the print very challenging to print due to the big flat surface and 2) it will not survive any impacts from diabolos. I was thinking about constructing a holder for the paper target and a jig for the sensors.

mjbedford, 2021-01-31:
Alan

Yes that you be great send me the PayPal link 

Martyn Bedford
+44 (0)7796692467

AzmodanRO, 2021-01-31:
> I am also going to construct parts for 3D printing as soon as I have the hardware, but I would suggest not to integrate the front plate: 1) ist going to make the print very challenging to print due [...]

1) i agree
2) the face is mounted on rails behind a shield
[image]

of course the face plate could be smaller, but you need to link the paper holder and the sensor jig somehow, so they are aligned.

ten-point-nine, 2021-01-31:
I was thinking the same.  A small frame that holds the sensors onto a stronger plate.

Allan

mjbedford, 2021-01-31:
I can easily change the frame to a square

It’s already modelled to fit to the back of a front plate made of other
material

It will be easier to align if it’s a square

Martyn Bedford
+44 (0)7796692467

>
> 
> I am also going to construct parts for 3D printing as soon as I have the
hardware, but I would suggest not to integrate the front plate: 1) ist
going to make the print very challenging to print due to the big flat
surface and 2) it will not survive any impacts from diabolos. I was
thinking about constructing a holder for the paper target and a jig for the
sensors.

ddraig576, 2021-01-31:
[image]
I was considering 4 corner mounts with a suitable sized template to get them equally distant similar to image, but if I read the earlier post right, does the centre need to be a set distance from the sensor? Or put it another way are the distances between sensors fixed? 

I like the idea of using light boxes as they already exist, but could design something to match the requirements of the sensor layout.

ten-point-nine, 2021-01-31:
The critical dimensions are

Face to face sensor distance. 
Perpendicularity
Orientation (45 degrees)

Face to face makes the target bigger
Perpendicularity skews the circle to an ellipse 
Orientation rotates the results

There is an override in my software to adjust the sensor angle, but it is fixed to 45 degrees.

Assuming everything is square and true, Alex’s software lets you adjust the centre.

Allan

ten-point-nine, 2021-01-31:
Yes 

The nose-to-nose distance is fixed by the 0 ring.  Root2 x 0 ring

There is a calibration in Alex’s software.

Allan

mjbedford, 2021-02-03:
could someone confirm the hole size in the sensor board and confirm that the replacing is still 3.80mm from the base of the sensor mount and 14.00 mm apart equidistant from the centre? I took these measurements from the PDF files in the cad.rar package. Thanks

ten-point-nine, 2021-02-03:
Martin 

The hole numbers look good although if you can hold off for a few days.  My PCB guy wants to move the connector to the front of the board, make it right angle, and polarize it.

The speaker and mounting holes will remain i the same place.

He is ahead of schedule on the version 3 boards so I’d you want I’ll move you from the version 2 list to 3.   The new boards have light control and face strike detection.

Allan

mjbedford, 2021-02-03:
Allan, Thanks, that sounds great

LotusElan, 2021-05-23:
Looking for confirmation.  I just received the version 3 board and sensors.  I am seeing conflicting information regarding the layout of the target box.  I see where the sensor are located at 12, 3 ,6 ,& 9 O'clock  and also skewed 45 degrees from that.  The 45 degree arrangement seems to be the most current info.  Is that correct?
Also, I want to confirm the component stack from front to back.
|| = Wood or whatever.     

**||**LED	Lights -------- Paper Target**||**Sensors ---------- Pellet Trap**||**

Is the distance from the microphone to the paper target important other than to avoid parallax?

Thanks
John

ten-point-nine, 2021-05-23:
Hi

Here is a link to the correct geometry.

sensor-geometry-1.pdf <https://allanbrownatl.files.wordpress.com/2021/04/sensor-geometry-1.pdf>

It shows the sensors 230 mm apart and rotated 45 degrees.

The 90 degree orientation was the first attempt and the sensors were too close.  

Can you tell me where the 90 degree documentation is so I can delete it.  Thanks

Cover
LED lights
Sensor
Paper
Trap

I have about a 35 mm space between the front face and the target plane.

Allan

LotusElan, 2021-05-23:
Thank you.

I looked up and saw your reply and thought how did I miss that?  Then I realized it was your really fast  reply.

The picture a few post above and the last page of the **freETarget - Interface Control Document**.

LotusElan, 2021-05-24:
Allan,

 

Thank you.

 

It is on the last page of the freETarget - Interface Control Document

 

John