ten-point-nine/freETarget #13: Need help!

Muammer06, 2021-02-20:
hey allan, I am student at university and our team make some machine learning things about 
shooting accuracy so we need record shootings like around 1000-1500. And I can just found your projects but I need help for setup. I can't understand to using which type of sensors and arduino coding. Also I am the new in coding like 3 months. if you help me thank you. I need this project in one month. thanks

AzmodanRO, 2021-02-20:
go to https://free-e-target.com/ to learn more about the project.

ten-point-nine, 2021-02-20:
Where are you located?

I have some detector boards I could send.

You would need for add your own processing boards.

Allan

Muammer06, 2021-02-21:
Hey, I am living in Ankara,Turkey. I am student at Hacettepe University and I am so glad about these boards. I have esp32 and arduino. Thank you again. Where we can talk my email is= muammert9@gmail.com

closed 2021-08-14 by AzmodanRO