ten-point-nine/freETarget #15: MainBoard PCB

iliahz, 2021-05-07:
I have the opportunity and really want to implement this project.
Please add the missing files in freETarget/KiCad/Main Board/:
Electronic Target-pcb
Electronic Target_net
From my side, I can adapt this project for use on SMT Nucleo Board with greater performance(https://www.st.com/en/evaluation-tools/stm32-nucleo-boards.html)
BRG

ten-point-nine, 2021-05-07:
Unfortunately you will need to manually enter the schematic for your particular board and do a layout to generate the appropriate Gerber files.

iliahz, 2021-05-07:
SMT Nucleo=Arduino
Can you provide these missing files?
Unfortunately, I don't have a lot of development experience. I'll just spend a lot of time learning and designing.
Experience: with Arduino about 10  working projects  in production +C# developer. 
my project: https://www.youtube.com/watch?v=y7md_O7nt5M 
my dream: electronic scoring systems + shooter training systems in one system.

ten-point-nine, 2021-05-07:
The version 3 schematics are on git hub.  You will need to generate gerbers if you need them.

Otherwise I can sell you circuit boards and you fly wire them to the NXP board

Allan

closed 2021-08-14 by AzmodanRO

Stefano-61, 2021-08-23:
Hello everyone, I introduce myself, I'm Stefano and I live in Italy near Treviso. I am a service technician. I travel for work also abroad. I am a fan of electronics and automation. As a hobby I like to shoot with the 10 m compressed air rifle. and 22 gauge rifle at 50 mt.
I found this project fantastic. I personally use EasyEDA for my PCBs. Now I try to create a PCB for the motherboard. I hope it will be of interest.
Good job everyone