ten-point-nine/freETarget #17: Doubt other type of plates

andressasousa, 2021-05-28:
[image]
[image]
[image]

Is it possible to use an Arduino Uno R3 SMD, a microcontroller board based on ATmega328, a Shield sensor Arduino 9 DOF? or in place of the Arduino 9 DOF Shield use a Ky038 Sound Sensor Module Arduino Microphone Sound Detector ?

andressasousa, 2021-05-28:
Is it possible to use an Arduino Uno R3 SMD, an ATmega328 based microcontrolled board, an Arduino 9 DOF Shield sensor? or, in place of the Arduino 9 DOF Shield, use a Ky038 sound sensor module, Arduino microphone, sound detector?

ten-point-nine, 2021-05-28:
Unfortunately no.

The Arduino typically samples around 40kHz for all channels.  This means that the four microphones would sample at 10kHz each.  

The freETarget samples at 8MHz per channel or 32MHz overall.  This is needed so as to resolve the position to well under 1mm which is what is needed for ISSF rules.

The freETarget board also supports the power supplies so that the pellet sound is picked up correctly.  The board also supplied the light and motor controls.

Allan

andressasousa, 2021-05-28:
Could you enter the name of the sensors used and the Arduino?

So that I can check if you are in my country.

ten-point-nine, 2021-05-28:
All of the information you need should be on GitHub already.

Look under the kicad folder 

Allan

ten-point-nine, 2021-05-28:
A complete hardware set is available from me for $200+$25 shipping

Allan

closed 2021-08-14 by AzmodanRO