ten-point-nine/freETarget #21: Error starting Software

RcamiruagaJ, 2021-08-14:
I am trying to run last version (soft and firm) in Windows 10 but 32 bits, and give error of database (problem with SQLite).
In spite off the soft run but I can not save the session.
I will appreciate help to solve this. Thank you for comments.
[Doc1.docx](https://github.com/ten-point-nine/freETarget/files/6986505/Doc1.docx)

AzmodanRO, 2021-08-14:
there is a file in the installation folder . subfolder "log". name "freETarget.<date>.log
can you attach here the most current file (for the date when you encountered this problem)

thank you

RcamiruagaJ, 2021-08-14:
The subfolder "log" is not created

AzmodanRO, 2021-08-14:
can you send a screenshot of the installation folder? where the freetarget.exe is located

thanx

AzmodanRO, 2021-08-14:
also, after you click OK in the "database problem" error window, click in the textbox that is barely visible between the target and the timer (there you can see the word "Log").
then press Ctrl-A, then press Ctrl-C, and then go to the github page (here) and press Ctrl-V to paste the content of that textbox. and send it

thanx

RcamiruagaJ, 2021-08-14:
OK here attached in v
[freETarget data files.docx](https://github.com/ten-point-nine/freETarget/files/6986622/freETarget.data.files.docx)
ersion 2 and in version 1.16 with the corresponding behaviour and logs

RcamiruagaJ, 2021-08-14:
Thank you so much for your support

AzmodanRO, 2021-08-14:
you have some write permission problems. i think your account is too restricted and the program cannot write to the installation folder or the database folder. i don;t know how you managed that :)

you need to execute the program as administrator. 
right-click on the freetarget icon, and from the context menu choose "Run as administrator".  and enter the administrator password if promted.

RcamiruagaJ, 2021-08-14:
Dear Sir,

Finally I run the software in a PC of 64 bits, I can not run in 32 bits.

After some tests I have noted that impacts in target are not the same that show in the display (software), I assume that is a calibration problem.

I have tested with "setting trigger point" document, all possible combinations... no good result.

And also I have tested setting the sensor positions (by software)...no good result.

My building is exactly to 230 mm and 165 mm between sensors.

I would be very appeciated if you can help me  on this subject.

Ramón

ten-point-nine, 2021-08-15:
What is the trip point set to?

Please send photos of target and display.

The sensors are 230 mm apart?

Allan

ten-point-nine, 2021-08-15:
What affects the accuracy is

Sensors in the correct location

Sensors 230 mm apart on the a diagonal

Trip point set to 1.25 volts

Paper close to the sensors

Remove face sensor 

Shoot five shots into a clean target and post the pc and target pictures.

Thanks

Allan

closed 2022-01-11 by AzmodanRO