ten-point-nine/freETarget #22: Wifi communication between target and PC

mjbedford, 2021-08-15:
I am working on a bridge between. TCP and serial COM as  a short term solution to using the WiFI accessory without changes to the PC Software which I believe will be significant.

In the long run, I would like to suggest a switch to message-based communication, this would remove a lot of the complexities of handling byte streams.  At the same time, I think it would be good to add support for connection to an external WiFi AP so that multiple targets can share the same network, again this would simplify the setup and make it possible to easily support extra UIs for scoreboards, central data loggers etc...

There are a number of very easy to use message libraries for Arduino and C# that make it easier to send and receive data over TCP.

I know this is a large change but I think. it will make the system much more open to extension.

This is my speciality having worked in message-based integration for the best part of thirty years so I am happy to take on the bulk of the work.

Would it be ok to create a branch to use to work on this in the long run?

AzmodanRO, 2021-08-15:
this is an open source project. you can create a fork of the entire project and modify as much as you want

ten-point-nine, 2021-08-15:
What’s the difference between a fork and a branch?

AzmodanRO, 2021-08-15:
A repository code branch, like a branch of a tree, remains part of the
original repository. The code that is branched (main trunk) and the branch
know and rely on each other. Like a tree trunk's branch, a code branch
knows about the trunk (original code base) it originated from.

Branches are best used: when they are created as temporary places to work
through a feature, with the intent to merge the branch with the origin.

Unlike a branch, a fork is independent from the original repository. If the
original repository is deleted, the fork remains. If you fork a repository,
you get that repository and all of its branches.

Forks are best used: when the intent of the ‘split’ is to create a
logically independent project, which may never reunite with its parent.

closed 2022-01-11 by AzmodanRO