ten-point-nine/freETarget #24: message termination characters

mjbedford, 2021-08-15:
I am having a problem handling the messages received from the target, the first message after the connection is freETarget"3.03.0 July 24, 2021" preceded by 2 characters the \r\n and followed. by \r\n

The first shot is preceded by \r\n
 it is  discarded  but all subsequent shots are handled ok

Is the first shot string created differently or is there something different about how the connection message is terminated that leaves additional data in the buffer?

mjbedford, 2021-08-15:
I've just had a read through the code in compute_hit it looks like you do not send \r\n after the shot so it gets handled ok, but on the connection message you send \r\n before and after the message. so I suppose its being flushed from the buffer

ten-point-nine, 2021-08-15:
Originally the version string was added for console debugging and there are lots of /r/n s added to space out the text.

mjbedford, 2021-08-16:
Allan, I have made one small change to compute_hit, in send_score
To add a \n to the end of the string for the shot message,

sprintf(str_b, "%s}", str_a);

sprintf(str_b, "%s}\n", str_a);

The absence meant that the receiver never saw the first shot as there was no new line to show the end, the next shot was processed but flushed the first shot. 

I've tested this and it looks like the target software does not notice the change, and it makes the wifi2com bridge work as expected. If you are happy with this could you merge the change into the master for your next release?

ten-point-nine, 2021-08-16:
Happy to do it.  It will either be 3.3.7 or 3.4.0.  

Have you tried sending a configuration over WiFi?   I don’t have a means of doing that.  It *should* work

Allan

mjbedford, 2021-08-16:
I haven’t tried yet.

Version one handles target to PC but it would be possible to do it both
ways. I’ll take a look for version two

Martyn Bedford
+44 (0)7796692467

>
> 
> Allan, I have made one small change to compute_hit, in send_score
> To add a \n to the end of the string for the shot message,
>
> sprintf(str_b, "%s}", str_a);
>
> sprintf(str_b, "%s}\n", str_a);
>
> The absence meant that the receiver never saw the first shot as there was
no new line to show the end, the next shot was processed but flushed the
first shot.
>
> I've tested this and it looks like the target software does not notice
the change, and it makes the wifi2com bridge work as expected. If you are
happy with this could you merge the change into the master for your next
release?

—
or Android