ten-point-nine/freETarget #25: Missing PCB design

RespawnDespair, 2021-08-16:
Hi,

in issue #15 it is also mentioned, but for a different reason.
I recently discovered this great project, absolutely terrific work!

I have a lot of experience with Open Source projects, so i always run a sanity check whenever i want to dive into something new.
With this project that leads me to some issues:

- No true BOM for the electronic components. One can reverse engineer the schematics, but no shipping list is provided
- No PCB design for the main board

The issues above prevent newcomers from making a quick assesment of the project. I have no qualms about people ordering boards, but in a true Open Source project all files should be available. 

With that being said, I would like to volunteer my time to help get the documentation in order. I was/am heavily involved in the Open.HD project where i mostly did project management and documentation. [Open.Hd](https://github.com/OpenHD/Open.HD) [Docs](https://openhd.gitbook.io/open-hd/)

I would love to start the build of this system myself and document the steps and necessary components.

My day-time profession is Developer and i have lots of experience with C# and Arduino, so no issues there. 

Is any additional help appreciated and if so, where would my efforts be most needed?

ten-point-nine, 2021-08-16:
If you can create a BOM and gerbers they can be added to github.

ten-point-nine, 2021-11-08:
> Hi, any news about it ?

The updated schematics are on the web site.

Gary-77, 2021-11-08:
Thanks for it :)
I was speaking about the PCB design, will be great to have it such as 3d parts.

AussieMakerGeek, 2022-04-30:
@ten-point-nine  Such a shame that you gatekeep the PCB design for such a great project. 

While I understand that you have put considerable time effort into the code and the project, if you make it so that people without technical knowledge of PCB layouts and high level electronics knowledge have no chance of making it without buying your boards, it might as well be a commercial product.

Considering the project can not be used on a commercial range (as it's not certified) and targeting personal use (no pun intended), having this as an open source project is useless if it's only able to be actually built by people with considerable PCB layout skills without buying your boards. That's not really in the true spirit of open source.

Personally - I have the skills but I just can't be bothered spending weeks laying it all out, ordering PCB, getting it wrong, redesigning, etc, etc, to the extent, i can't be bothered with the project, as much as i'd love to make one.  If PCB design was available, i'd give it a go.

Why not sell DIY PCB kits at least?

ten-point-nine, 2022-04-30:
The schematics are on GitHub hub and people with the technical knowledge can make a pcb from it.

ten-point-nine, 2022-04-30:
This could be your opportunity to contribute

AussieMakerGeek, 2022-05-10:
Ok, I'll take this on.  I'm still interested in porting it to the ESP32 but that is going to take a longer and still trying to get my head around the maths of it.

I've been going over the v3.1 schematic and I have a couple of questions:

Why the two negative voltage generators?  Looking at the spec of the devices, they do the same thing but one is SMD and the other is through hole but they appear to be in parallel in the schematic anyway.

Any specific reason to use the LM75 over say a DS18B20?  LM75 +/- 2.0 deg accuracy, the DS18B20 is +/- 0.5.

Were you going to publish the V3.1 schematic kicad files?

ten-point-nine, 2022-05-11:
Can we switch this to ***@***.*** ***@***.***>?

Depending on price and availability one of the two negative converters is used.

No reason not to switch the op amps.

The KiCad shematics appear to be up-to-date.

Allan

AussieMakerGeek, 2022-05-11:
Not sure what you mean by the first comment - Do you mean swap to email comms?

Perfect - Makes sense

The LM75 is the temp sensor, not the op amp

The KiCad Schematics (actual SCH files) for the mainboard are only v3.0, not v3.1. Only the PDF is 3.1. Not much point designing a board that's already outdated.

ten-point-nine, 2022-05-11:
I’ll check the drawings again.  The change from 3.0 to 3.1 was to bring the high pass filters onto the main board.  I don’t thing the filters are actually needed but I left them just-in-case

Allan

joemaya03, 2024-03-16:
[image]
[image]
[image]
I have made the PCB Design using the V3.1 Schematics, I have some queries regarding the KiCad Schematics.

ten-point-nine, 2024-03-16:
> Screenshot.2024-03-16.231304.png (view on web)
> Screenshot.2024-03-16.231154.png (view on web)
> Screenshot.2024-03-16.231132.png (view on web)

What questions?I see that there are component changesAllanSent from my iPhone

joemaya03, 2024-03-17:
What is the function of PIN D9, D10, D11 and D12 of the Arduino Mega2560?
Why have you shorted PIN D23 and PIN A7 of the Arduino Mega2560?

Apart from the above questions, I have incorporated the RC Filters in the Main Board itself, can you please have a look ? if everything is in order I will add the gerbers on Github.
[Schematic.pdf](https://github.com/ten-point-nine/freETarget/files/14626569/Schematic.pdf)

ten-point-nine, 2024-03-18:
The spare lines come into the board in-case in they are needed.  They are not used in the software D9 to D12 is a header that allows for configurations to be entered into the software.A and B are used for the multifunction switchesAllanSent from my iPhone

AussieMakerGeek, 2024-03-18:
Why go to the effort of designing a PCB now when Allan is just about to release an [ESP32 version](https://github.com/ten-point-nine/freETarget/issues/30#issuecomment-1930430719)?

ten-point-nine, 2024-03-19:
Thanks for the work.It helps people in some clubs and countries obtain electronic targets at the lowest possible cost.AllanSent from my iPhone

ten-point-nine, 2024-03-19:
> The PCB design took just 6 hours to complete. It was no trouble at all. I would like to contribute more, let me know if I can be of any assistance.
> And can you tell me which pin is A, B, Diag and Cal ?

8 pin debug headerCal 1A.   2B.   3Debug 4Sent from my iPhone