ten-point-nine/freETarget #29: 50m target not loading

thirdwheel1, 2022-01-10:
when I click the tab I get the error message - Unhandled exception has occurred.

and in the details it shows:
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at freETarget.Session.createNewSession(Event eventType, String user)
   at freETarget.frmMainWindow.initNewSession()
   at freETarget.frmMainWindow.disconnect()
   at freETarget.frmMainWindow.frmMainWindow_FormClosing(Object sender, FormClosingEventArgs e)
   at System.Windows.Forms.Form.OnFormClosing(FormClosingEventArgs e)
   at System.Windows.Forms.Form.WmClose(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4420.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
freETarget
    Assembly Version: 3.3.0.0
    Win32 Version: 3.3.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/freETarget/freETarget.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4400.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4360.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4390.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4190.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4390.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Windows.Forms.DataVisualization
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4200.0
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.DataVisualization/v4.0_4.0.0.0__31bf3856ad364e35/System.Windows.Forms.DataVisualization.dll
----------------------------------------
System.Data.SQLite
    Assembly Version: 1.0.112.1
    Win32 Version: 1.0.112.1
    CodeBase: file:///C:/Program%20Files%20(x86)/freETarget/System.Data.SQLite.DLL
----------------------------------------
System.Data
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4270.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_32/System.Data/v4.0_4.0.0.0__b77a5c561934e089/System.Data.dll
----------------------------------------
System.Transactions
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_32/System.Transactions/v4.0_4.0.0.0__b77a5c561934e089/System.Transactions.dll
----------------------------------------
System.EnterpriseServices
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_32/System.EnterpriseServices/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.EnterpriseServices.dll
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

AzmodanRO, 2022-01-10:
hi
can you upload the log file as well? 
it is located in the My Documents\freETarget\log\freETarget.<date>.log

thank you

thirdwheel1, 2022-01-10:
Thank you alex for your efforts, here is the log file, I just tried to load the 50m target again so hope this has captured it.

2022/01/10 17:05:11:824 | ------------------------------------------------------------------------

Starting freETarget - v3.3.0 ...
2022/01/10 17:05:12:105 | Template database version = 3   -   User database version = 3
2022/01/10 17:05:12:105 | User database at: C:\Users\third\Documents\freETarget\database
2022/01/10 17:05:12:386 | ### New Session 'Pistol Practice' started ###
2022/01/10 17:20:44:317 | ------------------------------------------------------------------------

Starting freETarget - v3.3.0 ...
2022/01/10 17:20:44:426 | Template database version = 3   -   User database version = 3
2022/01/10 17:20:44:442 | User database at: C:\Users\third\Documents\freETarget\database
2022/01/10 17:20:44:708 | ### New Session 'Pistol Practice' started ###
2022/01/10 17:21:13:494 | Error connecting: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.10.9:1090
2022/01/10 17:21:59:053 | ------------------------------------------------------------------------

Starting freETarget - v3.3.0 ...
2022/01/10 17:21:59:193 | Template database version = 3   -   User database version = 3
2022/01/10 17:21:59:193 | User database at: C:\Users\third\Documents\freETarget\database
2022/01/10 17:21:59:468 | ### New Session 'Pistol Practice' started ###
2022/01/10 17:22:02:617 | TCP channel open...
2022/01/10 17:22:03:367 | ****************
Connected to freETarget "3.04.8 September 23, 2021" on TCP = 192.168.10.9:1090
2022/01/10 17:22:03:401 | ### New Session 'Pistol Practice' started ###
2022/01/10 17:22:03:432 | freETarget "3.04.8 September 23, 2021"
2022/01/10 17:22:04:808 | {
"NAME":"1", 
"ANGLE": 45, 
"CALIBREx10": 45, 
"DIP": 0, 
"LED_BRIGHT": 99, 
"MFS": 0, 
"NAME_ID": 1, 
"PAPER_ECO": 0, 
"PAPER_TIME": 30, 
"POWER_SAVE": 30, 
"SEND_MISS": 0, 
"SENSOR": 230.00, 
"SN": 0, 
"STEP_COUNT": 0, 
"STEP_TIME": 0, 
"TARGET_TYPE": 0, 
"TEST": 0, 
"Z_OFFSET": 8, 
"NORTH_X": 0, 
"NORTH_Y": 0, 
"EAST_X": 0, 
"EAST_Y": 0, 
"SOUTH_X": 0, 
"SOUTH_Y": 0, 
"WEST_X": 0, 
"WEST_Y": 0, 
"IS_TRACE": 0, 
"TEMPERATURE": 7.00, 
"SPEED_SOUND": 0.34, 
"V_REF": 1.09, 
"TIMER_COUNT":5568, 
"DIP_HEX": 0x0, 
"WiFi": 1, 
"VERSION": "3.04.8 September 23, 2021", 
"BD_REV": 3.00 
}
2022/01/10 17:25:28:065 | *********
Disconnected.
2022/01/10 18:18:19:567 | ------------------------------------------------------------------------

Starting freETarget - v3.3.0 ...
2022/01/10 18:18:19:692 | Template database version = 3   -   User database version = 3
2022/01/10 18:18:19:692 | User database at: C:\Users\third\Documents\freETarget\database
[... 164 lines cut ...]
"STEP_COUNT": 0, 
"STEP_TIME": 0, 
"TARGET_TYPE": 0, 
"TEST": 0, 
"Z_OFFSET": 8, 
"NORTH_X": 0, 
"NORTH_Y": 0, 
"EAST_X": 0, 
"EAST_Y": 0, 
"SOUTH_X": 0, 
"SOUTH_Y": 0, 
"WEST_X": 0, 
"WEST_Y": 0, 
"IS_TRACE": 0, 
"TEMPERATURE": 16.50, 
"SPEED_SOUND": 0.34, 
"V_REF": 1.09, 
"TIMER_COUNT":5568, 
"DIP_HEX": 0x0, 
"WiFi": 1, 
"VERSION": "3.04.8 September 23, 2021", 
"BD_REV": 3.00 
}
2022/01/10 18:24:09:438 | *********
Disconnected.

AzmodanRO, 2022-01-10:
no.. the error is not in the log. if the error happened yesterday, you can find yesterday's log file in that directory.
also, you can attach the file here, not copy paste the text within.

from the previous error, it looks like the error message appeared when you were trying to close the window... so that is a little weird.

can you describe (or remake) the steps with performed when adding the 50m rifle event? 
what name did you give to the event?
is the 50m rifle event showing in the vertical events tab when you start the application?

thirdwheel1, 2022-01-10:
Thank you alex.
It happens every time I hit the 50m tab so did it for that file before I pasted it to you, the error message wants me to quit the programme so I do. I followed your great video on the software and followed what was suggested to do to create an event, chose the name from the options list that is why I know the target is in there somewhere.
Do you think I should just download the latest software again and over write it and try and set it all up again, it might just have it's knickers in a bit of a twist. Ta again!

AzmodanRO, 2022-01-10:
hm. ok.
can you take 2 screenshots and post the pictures here? one of the "active events" tab and one with the "events" tab with the 50m rifle event selected? without connecting to the target. you can use the <snipping tool> from windows

reinstalling the software will not help.

thirdwheel1, 2022-01-10:
Alex here they are, hope they are there !!!

[image]

[image]

AzmodanRO, 2022-01-11:
all settings look to be ok.
i cannot reproduce the error on my system (with target simulator or actually connected to the target via both usb and wifi/tcp).

the next step would be to delete (actually rename) the database. go to the _MyDocuments\freETarget\database_ directory and rename the **Storage.db** file to something else.
and than restart the app, add the 50m rifle event again, add it to the active events list, close the app, start the app, connect to the target and switch to the 50m rifle tab

thirdwheel1, 2022-01-11:
WOOOOOO HOOOOOOO you are a flaming genius - but you might know that already. 
I'm not really deep down computer savvy but I get by at the surface level so your help was followed correctly and it works, I down loaded a db reader but was not needed DOH just changed name of the root file and all is good. I owe you lots of beer.
Thank you so much for your help.

AzmodanRO, 2022-01-11:
you're welcome

still not clear how you managed to screw the database up... :)

closed 2022-01-11 by AzmodanRO

thirdwheel1, 2022-01-11:
Give me a computer and I can screw up anything!!!
One more question and then I will be out of your hair, I'm going to build a 50m target as well as the 10m but the target is slightly larger so I will need to move the sensors out a bit so they are behind the steel and plywood shield, in the software you can set the distance between sensors, so asking the obvious can I just change it to what I want and all is good?
George

AzmodanRO, 2022-01-11:
theoretically yes.
practically...i;ve tested with my target in changing the sensor diameter with 5mm increments, and did not see any variation in the results. 
this is something the arduino code is handling. the PC software just sends that parameter to the target. i;ve talked to Allan about this, but he could not explain it. (and neither can i). 

so i would encourage you to test, with airpistol. change the sensor diameter from 200 to 250 (in 10mm increments) and shoot a target (so with the normal paper roll and paper mask, but also a normal paper target in the back as control) and compare the results.

these are my results from 10 October:
[image]
[image]
[image]