ten-point-nine/freETarget #32: Mobile client

geniushkg, 2022-11-01:
I am software developer and so far I understand that all data passed in JSON format to windows application , not sure but with some technical guidance I think we can eliminate laptops completely and use Mobile device to display output.

would more then happy to write open source mobile app to connect to Free-target, but lets gauge intrest of community 
if you all think this feature will be helpful please plus one this issue , or add comment

AzmodanRO, 2022-11-01:
hi
the pc client was done because of the quick and easy work with .NET framework. if you want to write a mobile client, go ahead. it would be a great idea and a great addition to the project.

ten-point-nine, 2022-11-01:
Hello

Arjen in the Netherlands has been working on a solution for some time now.

Maybe you can get with him to work on it.

Allan