ten-point-nine/freETarget #33: The problem with the launch

digitalsound3312, 2023-01-28:
I designed the board and filter according to the scheme provided in EasyEDA, I can provide the source files with your permission. But only when I achieve working condition. At the moment, after the power supply, the X LED lights up. (S sensor error, probably). I ask for help, where to look? According to the oscilloscope, it can be seen from the control points that the sensors are all triggered. For each separately, it can be seen that after hitting the contacts 3,6,8,11 74HC00, 8 MHz strobing begins on the counter 74LV8154. Which suggests that the sensor, amplifier and trigger are working.
[image]
[image]

digitalsound3312, 2023-01-28:
I will add that I do not connect the Face sensor. I put a jumper on the S sensor. When hitting the I sensor, the terminal writes: PASS

MikeHoed, 2023-01-29:
experiencing exact the same issue. designed pcb according 3.1 schematics in EasyEDA and only x-LED.  high pass connected, but no face sensor connected. shot not registered.

ten-point-nine, 2023-01-29:
It is showing that the calibration is not completed.Adjust the trip voltage to 1.25 voltsTest that the sensors work Type ! To acknowledge the test completeSent from my iPhone

digitalsound3312, 2023-01-29:
Совершенно верно, я методом научного тыка нажал кнопку CALL в диалоговом окне, постучал по поверхности некоторое времяи заметил, что плата заработала. Мой вариант платы получился рабочим, если разрешат авторы, могу прислать исходные файлы, по которым сразу отправить на изготовление.