ten-point-nine/freETarget #36: Typo mismatch

rsamanez, 2024-09-24:
Hi,
Actually, I'm not sure if this is a typo or the code was modified for some reason.
It's in the Arduino software, in the "freeETarget.ino" file, in the arm() implementation function, line 380
```
if ( sensor_status & TRIP_EAST )
```
I think it should be changed with
```
if ( sensor_status & TRIP_NORTH )
```

ten-point-nine, 2024-09-24:
Thank you I’ll look into it.

Did it cause an error, or how did you find it?

Thanks

Allan

ten-point-nine, 2024-09-24:
You are correct it is a typo.Fortunately it doesn’t affect the normal operation of the target.I’ll see if it was copied to Version 5ThanksAllanSent from my iPhone

ten-point-nine, 2024-09-24:
ThanksI checked version 5 and that portion does the test a different way, so the bug was not propagated AllanSent from my iPhone